venari/timelapse #4: Rescue pi that's shutting down too quickly!

leighghunt, 2022-01-05:
As of 53086600fb3d9bccdf444562328c112366f83cee, the pi is shutting down too quickly to be able to SSH into it.

@ethandhunt can you check how to access the filesystem on the SD card on another machine so we can tweak the script.

Eventually we need to either check an API for tasking info, including being able to change start up frequency and duration, or maybe initially we have two times during the day that the pi stays on for say 5-10 minutes to enable us to log on.

leighghunt, 2022-01-05:
@ethandhunt can you make a quick note of how you solved this?

ethandhunt, 2022-01-05:
I got the sd card out of the pi and plugged it into my computer using a micro sd card adapter.
After it was mounted I changed test.py so that it didn't shutdown directly after booting

closed 2022-01-05 by ethandhunt