venari/timelapse #60: Kernel messages on newly installed Pi

leighghunt, 2023-09-15:
Receiving following messages to terminal after a fresh install.

```
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.370607] Internal error: Oops - undefined instruction: 0 [#2] ARM

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.598586] Process bash (pid: 5061, stack limit = 0x4fe3ef21)2] ARM

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.605129] Stack: (0xdce65ec8 to 0xdce66000)mit = 0x4fe3ef21)

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.611658] 5ec0:                   00000000 2756cd00 0001fb4a 00000000 000000c5 c2fa121a

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.618221] 5ee0: c755c800 c4f21380 ffffffff c38793a8 00000cc0 000001e1 001e1000 001e1000

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.618221] 5ee0: c755c800 c4f21380 ffffffff c38793a8 00000cc0 000001e1 001e1000 001e1000

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.624736] 5f00: 00000a55 c1af8000 c1af8000 174de3cf 00000000 cbb56f38 c89f5784 c4f213b0

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.624736] 5f00: 00000a55 c1af8000 c1af8000 174de3cf 00000000 cbb56f38 c89f5784 c4f213b0

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.631269] 5f20: 00000000 c2fa121a 0000081f dce65fb0 c4f21380 001e1d68 0000081f 00000255

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.631269] 5f20: 00000000 c2fa121a 0000081f dce65fb0 c4f21380 001e1d68 0000081f 00000255

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.657248] 5fa0: ffffffff c4f6a080 00c5387d c0008f94 0016b388 00000021 001e1d60 00165990

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.663736] 5fc0: 0016b378 00000030 b6eaa778 00000000 0015ae60 b6eaa744 b6eaa750 0016b388

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.670202] 5fe0: 0016b388 beb78a20 b6dcc0f0 b6dcbf24 80000010 ffffffff 00000000 00000000
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.676630]  wp_page_copy from 0x1e1000p 15 14:58:01 ...
 kernel:[  460.644292] 5f60: 2756cd00 c0c34c7c 0000081f c08cd220 001e1d68 dce65fb0 b6eaa744 b6eaa750

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:01 ...
 kernel:[  460.676630]  wp_page_copy from 0x1e100000000 00000000 00000003 c2fa121a b6dcbf24 80000010

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.596905] Internal error: Oops - undefined instruction: 0 [#3] ARM21a b6dcbf24 80000010

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.596905] Internal error: Oops - undefined instruction: 0 [#3] ARM021 001e1d60 00165990

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.819027] Process stat (pid: 5134, stack limit = 0x29115f0d) b6eaa744 b6eaa750 0016b388

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.819027] Process stat (pid: 5134, stack limit = 0x29115f0d) ffffffff 00000000 00000000

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.825600] Stack: (0xdce91ec8 to 0xdce92000)
 @  11 0:-*                                                                                                                                                                                                                                                                                  7m 2.36 1.0GHz 429M40% 59G4% 2023-09-15 14:58:03
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.825600] Stack: (0xdce91ec8 to 0xdce92000)

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.832171] 1ec0:                   00000812 c0c2f02c 00000073 c66828c0 c3b54c40 00000002

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.832171] 1ec0:                   00000812 c0c2f02c 00000073 c66828c0 c3b54c40 00000002

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.838762] 1ee0: c1a35100 c3b54c40 ffffffff c250a5a0 00100cca 00000001 b6fc8000 b6fc8000

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.845398] 1f00: 00000a54 c6762db8 c6762db8 00000000 00000000 00000000 c77c9724 c3b54c70

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.838762] 1ee0: c1a35100 c3b54c40 ffffffff c250a5a0 00100cca 00000001 b6fc8000 b6fc8000

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.845398] 1f00: 00000a54 c6762db8 c6762db8 00000000 00000000 00000000 c77c9724 c3b54c70

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.851989] 1f20: 00000000 c4441a99 00000017 dce91fb0 c3b54c40 b6fc8f20 00000017 00000254

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...                                                                                                                                                                                                                          7m 2.36 1.0GHz 429M40% 59G4% 2023-09-15 14:58:04
 kernel:[  463.858522] 1f40: c250a5a0 00000007 c3b54c70 c08cd358 dce91f4c c4441a99 00000008 00000812
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.858522] 1f40: c250a5a0 00000007 c3b54c70 c08cd358 dce91f4c c4441a99 00000008 00000812

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.865093] 1f60: b6fc8000 c0c34bfc 00000017 c08cd220 b6fc8f20 dce91fb0 00000000 bed87b70
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.865093] 1f60: b6fc8000 c0c34bfc 00000017 c08cd220 b6fc8f20 dce91fb0 00000000 bed87b70

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.871628] 1f80: bed87cfc c00189fc 00000002 00000003 00000001 bed87b88 b6fd1b18 a0000010
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.871628] 1f80: bed87cfc c00189fc 00000002 00000003 00000001 bed87b88 b6fd1b18 a0000010

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.878134] 1fa0: ffffffff c66c8000 00c5387d c0008f94 b6fc9028 00000000 b6fc8f20 00011f20

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.884673] 1fc0: b6ff9000 b6fb7000 bed87ba0 b6fc902c 00000003 00000000 bed87b70 bed87cfc

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.878134] 1fa0: ffffffff c66c8000 00c5387d c0008f94 b6fc9028 00000000 b6fc8f20 00011f20

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.884673] 1fc0: b6ff9000 b6fb7000 bed87ba0 b6fc902c 00000003 00000000 bed87b70 bed87cfc

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...                                                                                                                                                                                                                          7m 2.36 1.0GHz 429M40% 59G4% 2023-09-15 14:58:05
 kernel:[  463.891122] 1fe0: b6ffc880 bed87bf0 b6fd16d0 b6fd1b18 a0000010 ffffffff 00000000 00000000
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.891122] 1fe0: b6ffc880 bed87bf0 b6fd16d0 b6fd1b18 a0000010 ffffffff 00000000 00000000

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.897600]  filemap_map_pages from 0x0
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.897600]  filemap_map_pages from 0x0

Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.904138] Code: 73c5cb76 68737213 70f86db3 5fe469a4 (f6c517fa) 
Message from syslogd@sediment-pi-zero-w-v1-m at Sep 15 14:58:04 ...
 kernel:[  463.904138] Code: 73c5cb76 68737213 70f86db3 5fe469a4 (f6c517fa) 

```

leighghunt, 2023-09-15:
```
pi@sediment-pi-zero-w-v1-m:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.21+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 Mon Apr  3 17:19:14 BST 2023
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x0bc00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=B8:27:EB:61:3A:D7 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=4c4e106f-02 rootfstype=ext4 fsck.repair=yes rootwait
...
...
[   61.792826] NET: Registered PF_ALG protocol family
[   62.512257] rtc-ds1307 1-0068: registered as rtc0
[   62.513525] rtc-ds1307 1-0068: setting system clock to 2000-01-01T02:12:04 UTC (946692724)
[   64.925189] rtc-ds1307 1-0068: registered as rtc0
[   64.926493] rtc-ds1307 1-0068: setting system clock to 2000-01-01T02:12:06 UTC (946692726)
[   68.171031] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   69.513114] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   84.560743] usb 1-1: USB disconnect, device number 2
[   86.760611] Indeed it is in host mode hprt0 = 00021501
[   86.970550] usb 1-1: new full-speed USB device number 3 using dwc_otg
[   86.970794] Indeed it is in host mode hprt0 = 00021501
[   87.212193] usb 1-1: New USB device found, idVendor=05ac, idProduct=0267, bcdDevice= 8.51
[   87.212245] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   87.212264] usb 1-1: Product: Magic Keyboard
[   87.212278] usb 1-1: Manufacturer: Apple Inc.
[   87.212292] usb 1-1: SerialNumber: F0T739700Y1J1Y1AV
[   87.237709] apple 0003:05AC:0267.0004: fixing up Magic Keyboard battery report descriptor
[   87.239111] apple 0003:05AC:0267.0004: Fn key not found (Apple Wireless Keyboard clone?), disabling Fn key handling
[   87.239397] input: Apple Inc. Magic Keyboard as /devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.0/0003:05AC:0267.0004/input/input3
[   87.251062] apple 0003:05AC:0267.0004: input,hiddev96,hidraw0: USB HID v1.10 Keyboard [Apple Inc. Magic Keyboard] on usb-20980000.usb-1/input0
[   87.266493] input: Apple Inc. Magic Keyboard as /devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.1/0003:05AC:0267.0005/input/input4
[   87.332365] apple 0003:05AC:0267.0005: input,hiddev97,hidraw1: USB HID v1.10 Keyboard [Apple Inc. Magic Keyboard] on usb-20980000.usb-1/input1
[   87.347045] apple 0003:05AC:0267.0006: hiddev98,hidraw2: USB HID v1.10 Device [Apple Inc. Magic Keyboard] on usb-20980000.usb-1/input2
[  100.139449] Internal error: Oops - undefined instruction: 0 [#1] ARM
[  100.139770] Modules linked in: rtc_ds1307 regmap_i2c cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg bnep imx708 dw9807_vcm binfmt_misc tun brcmfmac vc4 nf_tables nfnetlink 8021q garp stp llc hci_uart btbcm bluetooth brcmutil ecdh_generic ecc libaes sha256_generic libsha256 snd_soc_hdmi_codec drm_display_helper cec cfg80211 drm_dma_helper drm_kms_helper hid_apple snd_soc_core bcm2835_unicam i2c_mux_pinctrl v4l2_dv_timings i2c_mux snd_compress snd_pcm_dmaengine bcm2835_codec(C) syscopyarea v4l2_fwnode sysfillrect sysimgblt v4l2_async bcm2835_v4l2(C) fb_sys_fops raspberrypi_hwmon v4l2_mem2mem bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops rfkill videobuf2_v4l2 videobuf2_common snd_bcm2835(C) videodev snd_pcm snd_timer i2c_bcm2835 snd mc vc_sm_cma(C) uio_pdrv_genirq fixed uio i2c_dev drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 [last unloaded: regmap_i2c]
[  100.142508] CPU: 0 PID: 822 Comm: bash Tainted: G         C         6.1.21+ #1642
[  100.142778] Hardware name: BCM2835
[  100.142911] PC is at linux_banner+0xc7a78/0x1dd784
[  100.143103] LR is at wp_page_copy+0x1e0/0x6ec
[  100.143282] pc : [<c0a0a1f4>]    lr : [<c01c9d50>]    psr: a0000113
[  100.143436] sp : dc8d5ec8  ip : c001cf7c  fp : c4fd13b0
[  100.143603] r10: 00000000  r9 : dc8d5fb0  r8 : c0c2f02c
[  100.143763] r7 : 00000000  r6 : 00000008  r5 : b6e97000  r4 : 00000000
[  100.143925] r3 : 00000098  r2 : c263d140  r1 : 00000000  r0 : 00000008
[  100.144142] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  100.144347] Control: 00c5387d  Table: 07550008  DAC: 00000055
[  100.144530] Register r0 information: non-paged memory
[... cut ...]
[  463.878134] 1fa0: ffffffff c66c8000 00c5387d c0008f94 b6fc9028 00000000 b6fc8f20 00011f20
[  463.884673] 1fc0: b6ff9000 b6fb7000 bed87ba0 b6fc902c 00000003 00000000 bed87b70 bed87cfc
[  463.891122] 1fe0: b6ffc880 bed87bf0 b6fd16d0 b6fd1b18 a0000010 ffffffff 00000000 00000000
[  463.897600]  filemap_map_pages from 0x0
[  463.904138] Code: 73c5cb76 68737213 70f86db3 5fe469a4 (f6c517fa) 
[  463.910533] ---[ end trace 0000000000000000 ]---
[  463.916898] note: stat[5134] exited with irqs disabled
```

leighghunt, 2023-09-15:
```
pi@sediment-pi-zero-w-v1-m:~$ uname -a
Linux sediment-pi-zero-w-v1-m 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
pi@sediment-pi-zero-w-v1-m:~$ cat /etc/o
opt/        os-release  
pi@sediment-pi-zero-w-v1-m:~$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

Possibly related to https://github.com/raspberrypi/linux/issues/5162?

Seems to happen regularly - perhaps every 10 minutes.

McMornan, 2024-03-08:
I second that - I did a fresh install with the bullseye image from raspberry foundation and get the same error.

McMornan, 2024-03-08:
Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.202791] Internal error: Oops - undefined instruction: 0 [#1] ARM

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204092] Process apt (pid: 2632, stack limit = 0x5052a68f)

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204126] Stack: (0xdc8cdec8 to 0xdc8ce000)

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204156] dec0:                   00000000 c004f950 00000000 c0211e34 00000004 dbb516c9

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204192] dee0: c266be00 c3c65a80 ffffffff c27ec5e8 00000cc0 00000c72 00c72000 00c72000

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204228] df00: 00000a55 c30c8030 c30c8030 13a063cf 00000000 cbad28d8 ca4181c8 c3c65ab0

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204263] df20: 00000000 dbb516c9 0000081f dc8cdfb0 c3c65a80 00c7208c 0000081f 00000255

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204298] df40: c27ec5e8 00000002 c3c65ab0 c08cd358 00000000 00000000 00000000 dbb516c9

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204332] df60: 00000000 c0c34c7c 0000081f c08cd220 00c7208c dc8cdfb0 bee6119c 00000000

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204367] df80: 00000000 c00189fc 00000000 00000006 00000aba b659f500 b6a72024 80000010

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204402] dfa0: ffffffff c6be6180 00c5387d c0008f94 00c72008 00000019 00cd1940 00c7200c

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204436] dfc0: 00cd4848 00000018 b6b4f744 00000002 00000000 bee6119c 00000000 00000000

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204470] dfe0: 00000005 bee610c0 00cd4850 b6a72024 80000010 ffffffff 00000000 00000000

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204514]  wp_page_copy from 0xc72000

Message from syslogd@Einkaufszettel at Mar  8 16:01:31 ...
 kernel:[166577.204591] Code: 73c5cb76 68737213 70f86db3 5fe469a4 (f6c517fa)

^C

McMornan, 2024-03-08:
pi@Einkaufszettel:~ $ uname -a
Linux Einkaufszettel 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
pi@Einkaufszettel:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@Einkaufszettel:~ $