venari/timelapse #85: Trouble getting cellular working on ESP32-S3-SIM7670G-4G

leighghunt, 2024-07-08:
End goal - get internet access over cellular.

I'm trying to establish a connection to the internet on the SIM7670G, but attempts so far have failed to get serial working.

Board: https://www.waveshare.com/wiki/ESP32-S3-SIM7670G-4G

Pinout definition: https://www.waveshare.com/esp32-s3-sim7670g-4g.htm

redhotchili0909, 2024-07-15:
Any success on this yet? Trying to attempt the same function

leighghunt, 2024-07-15:
> Any success on this yet? Trying to attempt the same function

Hi, replied separately on Reddit, but I should post skeletal code here too so I don't forget, and anyone else can find it :-)

Simple AT command code supplied from Waveshare support.

```
#include <Arduino.h>

static const int RXPin = 17, TXPin = 18;
static const uint32_t GPSBaud = 115200;

String rev;

void SentSerial(const char *p_char) {
  for (int i = 0; i < strlen(p_char); i++) {
    Serial1.write(p_char[i]);
    delay(10);
  }
  Serial1.write('\r');
  delay(10);
  Serial1.write('\n');
  delay(10);
}

bool SentMessage(const char *p_char, unsigned long timeout = 2000) {
  SentSerial(p_char);

  unsigned long start = millis();
  while (millis() - start < timeout) {
    if (Serial1.available()) {
      rev = Serial1.readString();
      if (rev.indexOf("OK") != -1) {
        Serial.println("Got OK!");
        return true;
      }
    }
  }
  Serial.println("Timeout!");
  return false;
}

void setup() {
  Serial.begin(115200);
  Serial1.begin(GPSBaud, SERIAL_8N1, RXPin, TXPin);

  while (!SentMessage("AT", 2000)) {
    delay(1000);
  }
  
  SentMessage("ATD10086;", 2000);
  SentSerial("ATE1;");
  SentSerial("AT+COPS?");
  SentSerial("AT+CGDCONT?");
  SentSerial("AT+SIMCOMATI");
}

void loop() {
  if (Serial1.available()) {
    rev = Serial1.readString();
    Serial.println(rev);
  }
}
```

leighghunt, 2024-07-19:
Might have it working now, as of dfd211e2622894f375397c69cc0dce1773f566d9 - need to do some more testing.

Areas I was confused by - looks like `modem.waitForNetwork()` is always hanging and returning false.

Additionally, I seem to have to call `modem.gprsConnect()`, which confused me during earlier testing, as I'm attempting to connect to 4G/LTE.

It is possible that this test result is actually using GPRS, which whilst better than nothing, is going to be pretty slow - I'll update this ticket as I find out more.

cc @redhotchili0909

leighghunt, 2026-08-18:
Migrating to [Lilygo T-SIM7670g-S3 board](https://lilygo.cc/products/t-sim-7670g-s3?variant=52420133486773) which appears to have better support & documentation than the Waveshare boards.

See #104

closed 2026-08-18 by leighghunt